dineshannayya/riscduino #3: PDK path error

Sya0, 2022-02-22:
I have installed pdk in a specific directory and I export it as `$PDK_ROOT`. However, when I run `make verify-..` test scenario, I get `/opt/pdk/sky130A not found, please install pdk before running make.` error. I changed the PDK_ROOT path, where in wb-port's Makefile, to absolute path. Then there was another error occured like below lines.
```
Include file libs.ref/sky130_fd_sc_hd/verilog/primitives.v not found
No top level modules, and no -s option.
```

dineshannayya, 2022-02-22:
Can you send me your Makefile run log

dineshannayya, 2022-02-22:
Can you replace ~  to absolute path .. As i am not sure as this command will run inside docker it can understand ~  location

Here is My Run log file for your reference
<pre>docker run -v /home/dinesha/workarea/opencore/git/riscduino_sram:/home/dinesha/workarea/opencore/git/riscduino_sram -v /opt/pdk/sky130A:/opt/pdk/sky130A \
                -v /home/dinesha/workarea/efabless/MPW-4/caravel:/home/dinesha/workarea/efabless/MPW-4/caravel \
                -e TARGET_PATH=/home/dinesha/workarea/opencore/git/riscduino_sram -e PDK_PATH=/opt/pdk/sky130A \
                -e CARAVEL_ROOT=/home/dinesha/workarea/efabless/MPW-4/caravel \
                -u : dineshannayya/dv_setup:mpw5 \
                sh -c &quot;cd /home/dinesha/workarea/opencore/git/riscduino_sram/verilog/dv/wb_port &amp;&amp; export SIM=RTL DUMP=OFF &amp;&amp; make&quot;
riscv64-unknown-elf-gcc -I /home/dinesha/workarea/efabless/MPW-4/caravel -march=rv32imc -mabi=ilp32 -Wl,-Bstatic,-T,/home/dinesha/workarea/efabless/MPW-4/caravel/verilog/dv/caravel/sections.lds,--strip-debug -ffreestanding -nostdlib -o wb_port.elf /home/dinesha/workarea/efabless/MPW-4/caravel/verilog/dv/caravel/start.s wb_port.c
riscv64-unknown-elf-objcopy -O verilog wb_port.elf wb_port.hex 
# to fix flash base address
sed -i &apos;s/@10000000/@00000000/g&apos; wb_port.hex
iverilog -g2005-sv -DFUNCTIONAL -DSIM -I /opt/pdk/sky130A \
-I /home/dinesha/workarea/efabless/MPW-4/caravel/verilog/dv/caravel -I /home/dinesha/workarea/efabless/MPW-4/caravel/verilog/rtl \
-I ../    -I ../../../verilog/rtl -I ../../../verilog \
-I ../../../verilog/rtl/yifive/ycr1c/src/includes    -I ../../../verilog/rtl/sdram_ctrl/src/defs -I ../../../verilog/rtl/i2cm/src/includes \
-I ../../../verilog/rtl/usb1_host/src/includes -I ../../../verilog/rtl/mbist/include \
wb_port_tb.v -o wb_port.vvp 
</pre>

dineshannayya, 2022-02-22:
Look like in your run PDK_PATH is not propagated to iverilog command 
In You run :  iverilog -g2005-sv -DFUNCTIONAL -DSIM -I  
Where as my run: iverilog -g2005-sv -DFUNCTIONAL -DSIM -I **/opt/pdk/sky130A**

Sya0, 2022-02-22:
Where does that **/opt/pdk/sky130A** path come from? My PDK_ROOT path something different. By the way, I run make commands with absolute path, I just changed them before paste here.

dineshannayya, 2022-02-23:
As there are continuous changes in PDK and caravel repo, I have copied the working PDK and caravel repo in /opt folder inside docker dineshannayya:mpw5 

I have update the docker with pdk/caravel repo at /opt folder.  You can re-pull the docker using 
>  docker pull dineshannayyya:mpw5

Also I have made  change in Repo makefile to pick PDK and caravel from docker, You can re-pull the git repo also

In Make file, I have removed PDK_PATH and CARAVEL_ROOT 

New Changed in run command:

$(DV_PATTERNS): verify-% : ./verilog/dv/% check-coremark_repo check-riscv_comp_repo check-riscv_test_repo
	docker run -v ${TARGET_PATH}:${TARGET_PATH} \
                -e TARGET_PATH=${TARGET_PATH}  \
                -u $(id -u $$USER):$(id -g $$USER) dineshannayya/dv_setup:mpw5 \
                 sh -c $(VERIFY_COMMAND)

If you want to over-ride the PDK/CARAVEL root, then you can update the makefile as below
$(DV_PATTERNS): verify-% : ./verilog/dv/% check-coremark_repo check-riscv_comp_repo check-riscv_test_repo
        docker run -v ${TARGET_PATH}:${TARGET_PATH} -v ${PDK_PATH}:${PDK_PATH} \
                -v ${CARAVEL_ROOT}:${CARAVEL_ROOT} \
                -e TARGET_PATH=${TARGET_PATH} -e PDK_PATH=${PDK_PATH} \
                -e CARAVEL_ROOT=${CARAVEL_ROOT} \
                 -u $(id -u $$USER):$(id -g $$USER) dineshannayya/dv_setup:mpw5 \
                 sh -c $(VERIFY_COMMAND)

Below test-case run faster as it uses only user_project_wrapper and wishbone to control the test case

make verify-user_basic
make verify-user_uart
make verify-user_risc_boot
make verify-user_spi
make verify-user_i2cm
make verify-user_uart_master

Below Test run take more time as it's full chip run with caravel + user_project_wrapper
make verify-wb_port
make verify-risc_boot
make verify-uart_master

If you want to dump, then you can add DUMP=ON swicth

closed 2022-02-24 by Sya0