dineshannayya/riscduino #4: Openlane implementation error

Sya0, 2022-02-25:
I updated my docker by replacing mpw5 in your docker repo. I got running simulation except spi that fails. However, when I run `make` command in openlane folder or `make user_project_wrapper` in riscdunio root path I got error about some path which I do not have.

```
Error: sta.tcl, 126 can't read "::env(CLOCK_PORT)": no such variable
[ERROR]: during executing: "openroad -exit /openlane/scripts/openroad/resizer.tcl |& tee >&@stdout /project/openlane/clk_skew_adjust/runs/clk_skew_adjust/logs/placement/7-resizer_design_optimization.log"
```

```
Run Directory: /project/openlane/clk_skew_adjust/runs/clk_skew_adjust                                                                                                                                                                      
Source not found.

Source: /project/openlane/clk_skew_adjust/runs/clk_skew_adjust/results/lvs/clk_skew_adjust.lvs_parsed.gds.log                                                                                                                              
Source not found.

[INFO]: check full report here: /project/openlane/clk_skew_adjust/runs/clk_skew_adjust/reports/final_summary_report.csv
[INFO]: Saving Runtime Environment
```

There is no path which begins with `project` in my directories. Is there any solution about it

dineshannayya, 2022-02-25:
I guess this may be due to openlane docker ..  

1.  Can you pull my project openlane docker:    riscduino/openlane:mpw5
> docker pull riscduino/openlane:mpw5

2. set the environmental variable
export IMAGE_NAME=riscduino/openlane:mpw5
export OPENLANE_IMAGE_NAME=riscduino/openlane:mpw5
export OPENLANE_TAG=mpw5

3. Run Make file
>   make user_project_wrapper

Note:  In the latest git version i have moved docker from my name to project name .. i.e now
dineshannayya/dv_setup:mpw5   =>   riscduino/dv_setup:mpw5
dineshannayya/openlane:mpw5   =>   riscduino/openlane:mpw5

I will cross-check if there any specific issue in SPI simulation

dineshannayya, 2022-02-25:
I have cross-check the spi simulation issue, look like one the .hex is not in git repo .. i have update it to git repo ..

You can update the repo .. this should resolve the simulation issue

Sya0, 2022-02-28:
I do not get it. I still get this path error. 
`couldn't read file "/home/sukruuzun/caravel_user_project/caravel/openlane/user_project_wrapper_empty/fixed_wrapper_cfgs.tcl": no such file or directory`.

My caravel example project (32 bit counter) openlane synthesis run flawless but this one gives me error about caravel. `fixed_wrapper_cfgs.tcl` script is under `user_project_wrapper`, not `user_project_wrapper_empty`.

dineshannayya, 2022-03-01:
If are using MPW-5 caravel repo .. there is change in fixed_wrapper_cfgs.tcl file location

In MPW-4 :  $::env(CARAVEL_ROOT)/openlane/**user_project_wrapper_empty**/fixed_wrapper_cfgs.tcl
in MPW-5:  $::env(CARAVEL_ROOT)/openlane/**user_project_wrapper**/fixed_wrapper_cfgs.tcl

You need to locally change this file path based on your caravel repo ...

Note: As there is continuous update in caravel repo,  My repo used still uses MPW4 caravel repo.

Sya0, 2022-03-01:
Yes, I use MPW-5 caravel repo. I do not want to make changes in caravel repo. Is there any other way, I made some changes Makefile but it does not work. Should I edit docker image ?

Sya0, 2022-03-01:
I changed them locally, now I have this;
```
FileNotFoundError: [Errno 2] No such file or directory: '/project/openlane/user_project_wrapper/../../lef//qspim.lef'
[ERROR]: during executing: "/openlane/scripts/mergeLef.py -i /project/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged_unpadded.lef /project/openlane/user_project_wrapper/../../lef//qspim.lef /project/openlane/user_project_wrapper/../../lef//pinmux.lef /project/openlane/user_project_wrapper/../../lef//wb_interconnect.lef /project/openlane/user_project_wrapper/../../lef//uart_i2cm_usb_spi.lef /project/openlane/user_project_wrapper/../../lef//wb_host.lef /project/openlane/user_project_wrapper/../../lef//yifive.lef /project/openlane/user_project_wrapper/../../lef//DFFRAM.lef -o /project/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged_unpadded.lef |& tee >&@stdout"
```

Sorry to bother you but I am new to this, I stuck a lot.

dineshannayya, 2022-03-01:
Default all the lef and gds files are compressed (.gz)  in git repo.

You need to unzip the file in lef and gds folder by using below command
gzip -d * 

OR If you using latest repo .. In the use below command in base folder
make unzip

Sya0, 2022-03-01:
I think, I fix the version issue, no more path or version error. However, I have pds error:
```
key "straps" not known in dictionary
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally
```

dineshannayya, 2022-03-01:
Are you using riscduino/openlane:mpw5 OR efabless/openlane  docker ?

I have done PDN hack in Openroad setup to get Power Straps only for vccd1/vssd1. Else in standard Openlane flow does power straps for all the power vccd1/vssd1/vccd2/vssd2/vdda1/vssa1/vdda2/vssa2
This should work with riscduino/openlane:mpw5 docker.

If you want to works with efabless/openlane docker .. Then you can mask below line in user_project_wrapper/Makefile
set ::env(PDN_CFG) $proj_dir/pdn_cfg.tcl

Sya0, 2022-03-01:
Yes, I use efabless/openlane. Is there any possible disadvantage of doing that hack during tape-out?
`Then you can mask below line in user_project_wrapper/Makefile` by meaning that should I comment that line ?

dineshannayya, 2022-03-01:
My Design IP uses only one set of power domain (vccd1/vssd1).  Default openlane flow does power straps for all power domain vccd1/vssd1/vccd2/vssd2/vdda1/vssa1/vdda2/vssa2, even though design need one set of power hook up for hard-macro. 
Due to these i need to keep more free space between each hard-macro to get all 8 vertical and horizontal power strip to get connected ..  As my design has less free space between IP.  I have hacked PDN script to excluded un-intended power straps generation, which helps me to keep only 2 vertical and horizontal power strip spacing between hard-ip.

Correction:  You need to comment pdn_cfg.tcl in user_project_wrapper/config.tcl to get openlane flow to working .. 
set ::env(PDN_CFG) $proj_dir/pdn_cfg.tcl

If openlane flow PDN connection breaks then you need to move around the macro location to get PDN to pass.

Sya0, 2022-03-01:
I passed lot of checks and synthesis including PDN but now I have this.
```
[ERROR]: during executing: "magic -noconsole -dnull -rcfile /home/sukruuzun/caravel_pdk/sky130A/libs.tech/magic/sky130A.magicrc /openlane/scripts/magic/gds_pointers.tcl </dev/null |& tee >&@stdout /project/openlane/user_project_wrapper/runs/user_project_wrapper/logs/finishing/20-gds_ptrs.log"                                                       
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal
```

dineshannayya, 2022-03-01:
@Sya0 I don't know if this issue related to latest openlane tools. you may need to get help from openlane team.

As you see some of the tool update breaks project .. periodically i check for working openlane docker and clone it to riscdino/openlane

closed 2022-03-04 by Sya0