dineshannayya/riscduino #7: MPW-5 single core git version

clp510, 2022-07-08:
How do I get your MPW-5 single core git version?
If i understand correctly, all your MPW code in the same git repo, same branch(master), How do I get different MPW version code? is that possible?

BRs
luping

dineshannayya, 2022-07-08:
Yes, Riscduino has manged through Single Git Repo for all the MPW project.  As there was continuous change in design, I was not clear on which version of the design will be finally picked in the MPW.  To understand the current design version, I have kept two register which indicate the last update and version id inside the pin mux block.  This register will help me to identify version of the chip manufactured.

Due to MPW lottery system , Riscduino-Dcore & Riscduino-QCore are part of MPW-5 shuttle and Riscduino Single core is not part of the MPW-5.
Is any specific reason why you are looking for MPW5 instead of latest version ?

clp510, 2022-07-09:
Thanks for your comments. and thanks for your good project.
Actually I am developing a new riscv core, and I'd like to pick up a stable code base to diverge, to continue the core side develop.
MPW5 is not a *must*, stable is more important for me .
And by the way ,another question is , you say riscduino is pin compatible with Arduino, but Arduino UNO is DIP package format which has only 28 pins, but the open MPW is QFN/WCSP package, It means,  I will draw my own PCB, not buy a arduino UNO , and replace the ATmega328 chip.

dineshannayya, 2022-07-09:
In MPW shuttle has fixed 38 Pin IO with  QFN/WCSP package for user project. I have tried to map 28 Pins as in arduino and rest of the pins are used for External FLASH (As MPW shuttle does have any on-chip FLASH).  You need to make your own PCB and bring the connector as in arudino. If in future if MPW support user defined own package + Internal FLASH, then we can make pin compatible ATMGA chip. 
If you planning to branch out, you may need to continuously monitor the changes as currently i am porting ardunio software driver and there will be some design fixes to match the ardunio functionality.

clp510, 2022-07-11:
Thanks to your comments, will follow your advice ~~

dineshannayya, 2022-07-11:
closing the ticket

closed 2022-07-11 by dineshannayya