dineshannayya/riscduino #11: openlane/Module call problem

pengpeng-lian, 2022-07-22:
When "make user_project_wrapper",  its config. tcl file does not contain all verilog files,  but only top and its next level. Why can it run all of them?

===================================code==================================
## Source Verilog Files
set ::env(VERILOG_FILES) "\
	$proj_dir/../../verilog/rtl//yifive/ycr1c/src/top/ycr_top_wb.sv \
	$proj_dir/../../verilog/rtl/user_project_wrapper.v"
### Black-box verilog and views

set ::env(VERILOG_FILES_BLACKBOX) "\
        $proj_dir/../../verilog/gl/qspim_top.v \
        $proj_dir/../../verilog/gl/wb_interconnect.v \
        $proj_dir/../../verilog/gl/pinmux.v     \
        $proj_dir/../../verilog/gl/uart_i2c_usb_spi_top.v     \
	$proj_dir/../../verilog/gl/wb_host.v \
	$proj_dir/../../verilog/gl/ycr_intf.v \
	$proj_dir/../../verilog/gl/ycr_core_top.v \
	$proj_dir/../../verilog/gl/ycr_iconnect.v \
	$proj_dir/../../verilog/gl/digital_pll.v \
	$::env(PDK_ROOT)/sky130A/libs.ref/sky130_sram_macros/verilog/sky130_sram_2kbyte_1rw1r_32x512_8.v \
	"
==========================================================================

=================================instruction==================================
make user user_project_wrapper
==========================================================================

pengpeng-lian, 2022-07-22:
Could it be because "set ::env(VERILOG_FILES_BLACKBOX) "\"?   BLACKBOX this parameter?

dineshannayya, 2022-07-22:
Each Macro are separately Hardened and Integrated as block box at user_project_wrapper.  If you include all the RTL at user_project_wrapper tool will not be able harden full design in one go.
You need to first run sub macro harden scripts and followed by user_project_wrapper final integration

make pinmux     
make uart_i2cm_usb_spi_top  
make wb_host          
make ycr_core_top   
make ycr_iconnect
make qspim_top  
make wb_interconnect  
make ycr_intf

make user_project_wrapper

pengpeng-lian, 2022-07-22:
Our team had questions, why separate integrations? Instead of using the total project which contains all the RTL code to integrate at once? Is there a flaw in doing this？

pengpeng-lian, 2022-07-22:
> Each Macro are separately Hardened and Integrated as block box at user_project_wrapper. If you include all the RTL at user_project_wrapper tool will not be able harden full design in one go. You [...]

 Sorry for the late reply, thank you very much for your answer.
Wouldn't it be more automatic to execute the "make master project" only once?
What is the purpose of this separate integration?

pengpeng-lian, 2022-07-22:
> Current Makefile flow does call automatically all the sub macro. I prefer to run the each macro on need basic to avoid long run time. As most of the time design changes are specific to one macro. For each top-level integration update, you don't need to re-harden all the block.

Thanks for the reply, very useful for me, really appreciate it, solved my problem!

closed 2022-07-22 by pengpeng-lian