dineshannayya/riscduino #13: When I try to open the def\lef file with <openroad>, I have problems

pengpeng-lian, 2022-08-01:
############instruction##############################
openroad> read_lef user_project_wrapper.lef

############problem##################################
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met2) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[WARNING ODB-0176] error: undefined layer (met3) referenced
[... 128 lines cut ...]
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_42) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_43) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_44) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_45) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_46) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_47) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_48) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_49) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_50) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_51) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_52) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_53) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_54) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_55) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_56) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_57) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_58) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_59) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_60) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_61) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_62) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_63) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_64) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_65) statement.
[WARNING ODB-0155] error: undefined site (unithd) referenced in row (ROW_66) statement.

dineshannayya, 2022-08-01:
I see openroad tool need unpaded lef. I am just added only user_project_wrapper.merged_unpadded.lef.gz in the lef directory
you can unzip and use this lef
[user_project_wrapper.merged_unpadded.lef.gz](https://github.com/dineshannayya/riscduino/files/9231818/user_project_wrapper.merged_unpadded.lef.gz)

You can use below command to open in openroad

read_lef user_project_wrapper.merged_unpadded.lef
read_def ../def/user_project_wrapper.def

pengpeng-lian, 2022-08-01:
> I see openroad tool need unpaded lef. I am just added only user_project_wrapper.merged_unpadded.lef.gz in the lef directory you can unzip and use this lef [...]

I followed your instructions once, but still had problems.
#############problem#######################
openroad> read_lef user_project_wrapper.merged_unpadded.lef
[INFO ODB-0222] Reading LEF file: user_project_wrapper.merged_unpadded.lef
[WARNING ODB-0218] error: VIARULE (L1M1_PR) undefined layer li1
[WARNING ODB-0218] error: VIARULE (L1M1_PR_R) undefined layer li1
[WARNING ODB-0218] error: VIARULE (L1M1_PR_M) undefined layer li1
[WARNING ODB-0218] error: VIARULE (L1M1_PR_MR) undefined layer li1
[WARNING ODB-0218] error: VIARULE (L1M1_PR_C) undefined layer li1
[WARNING ODB-0218] error: VIARULE (M1M2_PR) undefined layer met1
[WARNING ODB-0218] error: VIARULE (M1M2_PR_R) undefined layer met1
[WARNING ODB-0218] error: VIARULE (M1M2_PR_M) undefined layer met1
[WARNING ODB-0218] error: VIARULE (M1M2_PR_MR) undefined layer met1
[WARNING ODB-0218] error: VIARULE (M1M2_PR_C) undefined layer met1
[WARNING ODB-0218] error: VIARULE (M2M3_PR) undefined layer met2
[WARNING ODB-0218] error: VIARULE (M2M3_PR_R) undefined layer met2
[WARNING ODB-0218] error: VIARULE (M2M3_PR_M) undefined layer met2
[WARNING ODB-0218] error: VIARULE (M2M3_PR_MR) undefined layer met2
[WARNING ODB-0218] error: VIARULE (M2M3_PR_C) undefined layer met2
[WARNING ODB-0218] error: VIARULE (M3M4_PR) undefined layer met3
[WARNING ODB-0218] error: VIARULE (M3M4_PR_R) undefined layer met3
[WARNING ODB-0218] error: VIARULE (M3M4_PR_M) undefined layer met3
[WARNING ODB-0218] error: VIARULE (M3M4_PR_MR) undefined layer met3
[WARNING ODB-0218] error: VIARULE (M3M4_PR_C) undefined layer met3
[WARNING ODB-0218] error: VIARULE (M4M5_PR) undefined layer met4
[WARNING ODB-0218] error: VIARULE (M4M5_PR_R) undefined layer met4
[WARNING ODB-0218] error: VIARULE (M4M5_PR_M) undefined layer met4
[WARNING ODB-0218] error: VIARULE (M4M5_PR_MR) undefined layer met4
[WARNING ODB-0218] error: VIARULE (M4M5_PR_C) undefined layer met4
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (pwell) referenced
[WARNING ODB-0176] error: undefined layer (nwell) referenced
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (pwell) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (pwell) referenced
[WARNING ODB-0176] error: undefined layer (nwell) referenced
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (pwell) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (li1) referenced
[WARNING ODB-0176] error: undefined layer (met1) referenced
[WARNING ODB-0176] message limit reached, this message will no longer print
[INFO ODB-0225]     Created 451 library cells
[INFO ODB-0226] Finished LEF file:  user_project_wrapper.merged_unpadded.lef
[ERROR ODB-0292] LEF data from user_project_wrapper.merged_unpadded.lef is discarded due to errors
ODB-0292
while evaluating {read_lef user_project_wrapper.merged_unpadded.lef }

pengpeng-lian, 2022-08-01:
> I followed your instructions once, but still had problems. #############problem####################### openroad> read_lef user_project_wrapper.merged_unpadded.lef [INFO ODB-0222] Reading LEF file: [...]

##############success lef################################
openroad> read_lef lef/user_project_wrapper.merged_unpadded.lef
[INFO ODB-0222] Reading LEF file: lef/user_project_wrapper.merged_unpadded.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 25 technology vias
[INFO ODB-0225]     Created 451 library cells
[INFO ODB-0226] Finished LEF file:  lef/user_project_wrapper.merged_unpadded.lef
###########################################################

################failed def#####################################
openroad> read_def def/user_project_wrapper.def
[ERROR ORD-0005] No technology has been read.
ORD-0005
while evaluating {read_def def/user_project_wrapper.def }
########################################################

dineshannayya, 2022-08-01:
This works for me .. Not sure issue related to openroad tool version difference .. I have standalone installed openroad application

[image]

pengpeng-lian, 2022-08-01:
> This works for me .. Not sure issue related to openroad tool version difference .. I have standalone installed openroad application
> 
> [image]

，

###########merged_unpadded.lef######################################
Thank you very very much!
 I have solved the problem on my side, one last question, how to generate the unpaded lef file? .merged_unpadded.lef

dineshannayya, 2022-08-01:
merged_unpadded.lef are generated when you run openlane flow. For  user_project_wrapper , I see them generated at "openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged_unpadded.lef"

closed 2022-08-01 by pengpeng-lian