dineshannayya/riscduino #14: question about clock skew adjust

clp510, 2022-08-02:


clp510, 2022-08-02:
In your design I have seen a lot of clock skew adjust logic, why clock skew adjust is necessary ?

dineshannayya, 2022-08-02:
When two macro interact each other in same clock domain. To met timing closure we need to match the clock latency between these macro.  Typically in commercial tool will have option to define clock latency to be achieved during macro hardening. 
These feature is not available in openlane, So we are adding clock skew adjust logic to each entry point of clock tree of the macro.  During the Top-level timing analysis we will adjust each the macro clock skew inside each macro to achieve clean timing closure.  All clock skew adjust delay are controlled by register inside the wb_host block.

clp510, 2022-08-03:
I see, thanks a lot.

closed 2022-08-03 by dineshannayya