dineshannayya/riscduino #15: About the change of FP_SIZING from ABSOLUTE to RELATIVE

pengpeng-lian, 2022-08-03:
###############################################################
Hello dear professor, what changes do I need to be aware of when FP_SIZING changes from absolute to relative, apart from the config.tcl configuration? Things like sdc files etc

dineshannayya, 2022-08-03:
FP_SIZING=absolute helps, when you have multiple macro and you want to decide on how the Top-level floor plan will look. FP_SIZING=relative can help you to find minimum size can be achieved through the flow.  Then you may need to adjust the Macro placement value at user_project_wrapper level

pengpeng-lian, 2022-08-03:
> FP_SIZING=absolute helps, when you have multiple macro and you want to decide on how the Top-level floor plan will look. FP_SIZING=relative can help you to find minimum size can be achieved through the flow. Then you may need to adjust the Macro placement value at user_project_wrapper level

Thank you for your answer!

closed 2022-08-03 by pengpeng-lian