dineshannayya/riscduino #16: PDN probelm

pengpeng-lian, 2022-08-04:
I have now put all the Verilogs together to run a macro, but have encountered a problem, reporting the following error
#############ERROR###############################
[STEP 6]
[INFO]: Running Tap/Decap Insertion...
[INFO]: Connecting Power: vccd1 & vssd1 to All internal macros.
[INFO]: Generating PDN...
[ERROR]: PDN generation failed.
[ERROR]: You may need to adjust your macro placements or PDN  offsets/pitches to power all standard cell rails (or other PDN stripes)  in your design.
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'user_project_wrapper/runs/user_project_wrapper/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'user_project_wrapper/runs/user_project_wrapper/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

#########################config.tcl###############################################
set ::env(PDK) "sky130A"
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hd"

# YOU ARE NOT ALLOWED TO CHANGE ANY VARIABLES DEFINED IN THE FIXED WRAPPER CFGS 
source $::env(CARAVEL_ROOT)/openlane/user_project_wrapper/fixed_wrapper_cfgs.tcl

# YOU CAN CHANGE ANY VARIABLES DEFINED IN THE DEFAULT WRAPPER CFGS BY OVERRIDING THEM IN THIS CONFIG.TCL
source $::env(CARAVEL_ROOT)/openlane/user_project_wrapper/default_wrapper_cfgs.tcl

set script_dir [file dirname [file normalize [info script]]]
set proj_dir [file dirname [file normalize [info script]]]

set ::env(ROUTING_CORES) "6"

set ::env(DESIGN_NAME) user_project_wrapper
set verilog_root $proj_dir/../../verilog/

# User Configurations
#
set ::env(DESIGN_IS_CORE) 1
set ::env(FP_PDN_CORE_RING) 1

## Source Verilog Files
set ::env(VERILOG_FILES) "\
	$proj_dir/../../verilog/rtl//yifive/ycr1c/src/top/ycr_top_wb.sv                 \
	$proj_dir/../../verilog/rtl/user_project_wrapper.v                              \
    $script_dir/../../verilog/rtl/clk_skew_adjust/src/clk_skew_adjust.gv \
    $script_dir/../../verilog/rtl/pinmux/src/pinmux.sv     \
    $script_dir/../../verilog/rtl/pinmux/src/pinmux_reg.sv \
    $script_dir/../../verilog/rtl/pinmux/src/gpio_intr.sv  \
    $script_dir/../../verilog/rtl/pinmux/src/pwm.sv        \
    $script_dir/../../verilog/rtl/pinmux/src/timer.sv        \
    $script_dir/../../verilog/rtl/lib/pulse_gen_type1.sv   \
    $script_dir/../../verilog/rtl/lib/pulse_gen_type2.sv   \
    $script_dir/../../verilog/rtl/lib/registers.v          \
    $script_dir/../../verilog/rtl/lib/ctech_cells.sv     \ 
      
    $script_dir/../../verilog/rtl/lib/reset_sync.sv      \
    $script_dir/../../verilog/rtl/qspim/src/qspim_top.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_if.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_regs.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_fifo.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_clkgen.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_ctrl.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_rx.sv \
    $script_dir/../../verilog/rtl/qspim/src/qspim_tx.sv	\

    $script_dir/../../verilog/rtl/uart/src/uart_core.sv  \
    $script_dir/../../verilog/rtl/uart/src/uart_cfg.sv   \
    $script_dir/../../verilog/rtl/uart/src/uart_rxfsm.sv \
    $script_dir/../../verilog/rtl/uart/src/uart_txfsm.sv \
    $script_dir/../../verilog/rtl/lib/async_wb.sv   \
    $script_dir/../../verilog/rtl/lib/async_fifo.sv      \
    $script_dir/../../verilog/rtl/lib/async_fifo_th.sv   \
    $script_dir/../../verilog/rtl/lib/double_sync_low.v  \
[... 177 lines cut ...]
                        li1   150  750 833.1  1166.54,\
                        met1  150  750 833.1  1166.54,\
                        met2  150  750 833.1  1166.54,\
                        met3  150  750 833.1  1166.54,\	
	                met5  0 0 2920 3520"

set ::env(FP_PDN_POWER_STRAPS) "vccd1 vssd1 1, vccd2 vssd2 0, vdda1 vssa1 0, vdda2 vssa2 0"

###PDN offset and pitch
set ::env(FP_IO_VEXTEND) 4
set ::env(FP_IO_HEXTEND) 4

set ::env(FP_PDN_VPITCH) 100
set ::env(FP_PDN_HPITCH) 100
set ::env(FP_PDN_VWIDTH) 5
set ::env(FP_PDN_HWIDTH) 5

set ::env(FP_PDN_VOFFSET) "5"
set ::env(FP_PDN_HOFFSET) "10"

set ::env(FP_PDN_HORIZONTAL_HALO) "10"
set ::env(FP_PDN_VERTICAL_HALO) "10"

set ::env(FP_PDN_VSPACING) "15.5"
set ::env(FP_PDN_HSPACING) "10"

pengpeng-lian, 2022-08-04:
##########################Supplementary macro.cfg###########################
u_dcache_2kb                150             130            N
u_icache_2kb                950             130            N
u_tsram0_2kb                150             750            N

pengpeng-lian, 2022-08-04:
I checked the def and found the cells overlapped!

dineshannayya, 2022-08-04:
I am not tried flat design in openlane flow.  You need to raise issue in openroad/openlane forum.

closed 2022-08-05 by pengpeng-lian