dineshannayya/riscduino #17: Did you download the latest version of your riscduino, which is the latest openroad tool? Also I'm having some problems with make

pengpeng-lian, 2022-08-17:
I've had the same problem as you before, you asked it at openlane, see this URL.
##################URL###########################PDN--MACRO--HALO###############
···https://github.com/The-OpenROAD-Project/OpenLane/issues/1262···

I need to download the latest version of openroad tools, is it git pull your new version of riscduino and I'm good to go?
I get the following error when I make
#########error#######################
###############################################
sh: flow.tcl: command not found
make: *** [Makefile:44: user_project_wrapper] Error 127

dineshannayya, 2022-08-18:
This issue is resolved from me, You need to pick the latest openlane version

pengpeng-lian, 2022-08-18:
Hello! How do I update the openlane version on the riscduino？

dineshannayya, 2022-08-18:
In the latest commit i have pushed the openlane folder into my docker. Additionally you need to install PDK

pengpeng-lian, 2022-08-18:
I get the following error when I make, How to solve this problem？
###################################problem#######################
jplian@riscduino/openlane$ make user_project_wrapper
###############################################
sh: flow.tcl: command not found
make: *** [Makefile:44: user_project_wrapper] Error 127

dineshannayya, 2022-08-18:
Did you pick the latest riscduino git database ?
Are you using which docker ?  riscduino/openlane:mpw7 ?

Try with latest docker update:
docker pull  riscduino/openlane:mpw7

pengpeng-lian, 2022-08-19:
I've successfully updated docker, riscduino/openlane:mpw7, but I've encountered an openroad problem during the run, could you help me?
###################################
[INFO]: Running Detailed Placement (log: user_project_wrapper/runs/user_project_wrapper/logs/placement/9-detailed.log)...
**[ERROR]: during executing openroad script /openlane/scripts/openroad/dpl.tcl**
[ERROR]: Log: user_project_wrapper/runs/user_project_wrapper/logs/placement/9-detailed.log
[ERROR]: Last 10 lines:
[INFO ODB-0226] Finished LEF file:  /work/home/jplian/projects_riscduino/risduino_1/riscduino_1/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /work/home/jplian/projects_riscduino/risduino_1/riscduino_1/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/placement/7-global.buffers_removed.def
[INFO ODB-0128] Design: user_project_wrapper
[INFO ODB-0130]     Created 645 pins.
[INFO ODB-0131]     Created 4 components and 1351 component-terminals.
[INFO ODB-0132]     Created 8 special nets and 0 connections.
[INFO ODB-0133]     Created 974 nets and 1311 connections.
[INFO ODB-0134] Finished DEF file: /work/home/jplian/projects_riscduino/risduino_1/riscduino_1/openlane/user_project_wrapper/runs/user_project_wrapper/tmp/placement/7-global.buffers_removed.def
**Error: dpl_cell_pad.tcl, 1 can't read "::env(DPL_CELL_PADDING)": no such variable**
child process exited abnormally

pengpeng-lian, 2022-08-19:
My problem has been solved by setting the parameters

closed 2022-08-19 by pengpeng-lian