dineshannayya/riscduino #21: How can I program the RISCDUINO SOC?

Aman-ECE, 2022-09-05:
I saw this RISCDUINO SoC project on the efabless website and it blew up my mind so interestingly I landed upon your GitHub page. I have a set of questions in my mind which are as follows:

Q.1: Have you received and tested the RISCDUINO?
Q.2: How can I burn the program in RISCDUINO?
Q.3: What are the tools (compilers, IDE, etc) required to build the firmware? I've seen that you are working on integration with Arduino IDE but also fails to find any solution for windows. is there any other possible way to carry on the development with windows?

dineshannayya, 2022-09-05:
Q.1: Have you received and tested the RISCDUINO?
<Ans> RISCDUINO project successfully got a tape-out slot efabless in
MPW-3/MPW-4/MPW-5/MPW-6 Shuttle. Even though tape-out was done long back,
still efabless yet to deliver MPW-2 chip's itself. Only chip tape-out date
hanpening as per schedule, not the chip delivery.  So I have not yet
received my first version of RISCDUINO MPW-3 chip.
Currently I am refreshing my 5th version of RISCDUINO in MPW-7 which has
most of the Arduino Digital functionality.

Q.2: How can I burn the program in RISCDUINO?
<Ans> We have three ways to Burn the RISCDUINO,
       A. Using UART I/F as in Arduino upload function
       B.  Using SPI ISP function as in Arduino
       C.  Caravel has extra RISCV core, which can be used for flash
programming.
Q.3: What are the tools (compilers, IDE, etc) required to build the
firmware? I've seen that you are working on integration with Arduino IDE
but also fail to find any solution for windows. Is there any other possible
way to carry on the development with windows?
<Ans> We need RISCV tool set and these will be part of RISCDUINO board
plug-in. For User it will look like one more type of arduino board and need
to just link the board driver
https://github.com/dineshannayya/riscduino_board/
in the arduino-ide tool.  I have tested it from arudino-ide 11 version. I
expect it will be be transparent for windows/unix. But I have still
validated it in the unix/ubuntu version only.
I have ported and validated following arudino example in Riscduino
environment

   1.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_arrays/arduino_arrays.ino
   2.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_ascii_table/arduino_ascii_table.ino
   3.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_digital_port_control/arduino_digital_port_control.ino
   4.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_gpio_intr/arduino_gpio_intr.ino
   5.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_hello_world/arduino_hello_world.ino
   6.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_i2c_scaner/arduino_i2c_scaner.ino
   7.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_i2c_wr_rd/arduino_i2c_wr_rd.ino
   8.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_multi_serial/arduino_multi_serial.ino
   9.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_risc_boot/arduino_risc_boot.ino
   10.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_string/arduino_string.ino
   11.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_switchCase2/arduino_switchCase2.ino
   12.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_timer_intr/arduino_timer_intr.ino
   13.
   https://github.com/dineshannayya/riscduino/blob/master/verilog/dv/arduino_ws281x/arduino_ws281x.ino

I expect it will take at least one more year to get a working project.

Aman-ECE, 2022-09-07:
Thank you so much for the detailed answers and going forward with my interest in RISCDUINO,

Q.1 As your GitHub page (https://github.com/dineshannayya/riscduino_board/) says that the installation of RISCDUINO boards is supported for macOS and Linux. Along with the porting and validation of Arduino examples in the RISCDUINO environment, do you also working on making compatible the RISCDUINO boards for windows too? Currently, I am getting the following message in the Arduino IDE (v2.0 RC) when I tried it with windows.

[image]

Q.2 How are you testing and working on further developments? Had you gotten the RISCDUINO fabricated from some other services/company/programs other than efabless? or are you testing with some sort of simulation-based approach? In short, I am very much interested to see the RISCDUINO working in the real world so do you have this chip in your hands yet from any other source?

Aman-ECE, 2022-09-26:
> 

Hi

Waiting for your response...

dineshannayya, 2022-10-01:
@Aman-ECE Sorry busy with other technical work. 

A1. currently I have verified only in Ubuntu, I have moved this issue as bug. you can also contribute fix this issue as it's need Arduino board integration methodology. 
A2. I have not yet tried this design other than efabless MPW shuttle. Design has been validated with around 100+ RTL simulation.  For bring-up, we can use the Arduino board itself with just replacing ATMGA chip with this one.

dineshannayya, 2023-07-19:
I have integrated it to windows version now.

Aman-ECE, 2023-07-19:
Thanks for the update!!

closed 2023-07-19 by Aman-ECE