dineshannayya/riscduino #24: Discrepencies in the readme.

iamKarthikBK, 2023-07-18:
Hi.

Your readme says this:
Step-2: Clone , update the Submodule, unzip the content

```shell
   git clone https://github.com/dineshannayya/riscduino.git
   cd riscduino
   git submodule init
   git submodule update
   make unzip
```
but the commands don't do what is expected. has there been an update which has not been reflected in this readme ?
`make unzip` complains about `lef/*.gz` files being missing. Is there something im missing?

Thanks in advance!

dineshannayya, 2023-07-18:
Normally I remove lef/* gds/* folder from the repo to keep git repo smaller; As these folder are need only silicon tapeout only (You can mask these unzip function inside Makefile).

Any how, I have check-in these folder back, you can resync the repo.

Vinayakamk, 2023-07-19:
thanks ,for the reply sir,
but in verilog gl file is also broken while unziping.
as u said ,,whether i can  remove  gzip verliog from make file?

! [image](https://github.com/dineshannayya/riscduino/assets/137029135/c974db1c-b90e-496b-9a0d-aa1b37b5d388)

dineshannayya, 2023-07-19:
You can skip/bypass the zip/unzip step, there are only required for tapeout package release.