dineshannayya/riscduino #31: Iverilog Version

redpanda3, 2024-07-14:
[image] Hi Dinesh, I have a problem when verifying the user_usb, the iverilog I'm using gives me a warning like this, could you tell me which iverilog you are using?

redpanda3, 2024-07-14:
mine is 11.

dineshannayya, 2024-07-14:
These are warning, simulation still will work.

redpanda3, 2024-07-14:
[image]
[image]
Could you please take a look at it here? I see no waveform generated.

dineshannayya, 2024-07-14:
I don't see Error details, Mostly if will be at starting of the log

redpanda3, 2024-07-14:
[image]
Yes, you are correct. Seems that problem came from AES module, I will take a look at that. Thank you.

dineshannayya, 2024-07-14:
Looks like there are lint issues due to the latest iverilog, I have pushed
fixes back to repo .. You can pull the changes and sub-git repos.

redpanda3, 2024-07-14:
[image]
In the security_core provided in the submodule, the version might not be the latest(has no idle signal).  After git pull the latest and manually upgrade the security_core, dv now works. Thank you.